kartchnb/AutoTowersGenerator #118: Publish last version (2.6.1) on Cura on markteplace

brunetton, 2024-07-12:
Hi

Current plugin version on plugins marketplace is "2.6.0 - last update on 23-12-2022"
Last version available in [releases](https://github.com/kartchnb/AutoTowersGenerator/releases) is 2.6.1 

I think it's time to update marketplace version :)

lebajez, 2024-08-29:
trying to get it working with snap and deb package - both paths /snap/bin(openscad, /usr/bin/openscad are not valid
also removing the plugin restarting cura and reinstalling it seems not to solve this issue.

upgrading the marketplace version to the actual would be much appreciated !

Blauenfeldt, 2024-12-15:
Current version on Cura marketplace is 2.7.2

meier2009, 2024-12-18:
did i miss something or is the custom settings gone?

Was able to do every  tower or bed customized. as i have greater bed then 220x220 for excample

kartchnb, 2024-12-20:
That probably means I managed to screw up the OpenScad integration.  Again.  I'll try to take a look when I have a moment.

meier2009, 2024-12-20:
Oh, thx for the hint! 😎 

I looked for scad and did not find it, because was a new computer where scad was yet not installed! 

I did it, put the right path back and it worked... 🙈 

My fault! 😅
Sorry... 🙈 

Maybe give a hint, that scad is not installed, in the custom section, for dumb users  like me! 😎 🤣

kartchnb, 2024-12-20:
So glad you figured it out!  That's a relief!

I'll think about how best to add a hint, in case OpenSCAD is not installed.

Cheers!