kartchnb/AutoTowersGenerator #125: Confusing versions

mpwab, 2024-10-10:
Please explain which version I should download for Ultimaker Cura 5.8.1.
It's very confusing with all the numbers in the file names that don't match any Cura version.
Plugin from Marketplace is only 2.6.0
Thanks in advance

an0nymz, 2024-12-12:
I am also finding this very confusing to the point of being an issue. The readme says to "Download the .curapackage file that corresponds to your version of Cura", but I'm on version 5.9.0 and the files under the latest 2.7.2 version of this plugin start at 7.9.0 and only go up from there?

kartchnb, 2024-12-14:
Thanks for pointing this out. I normally update the file names in the releases to include the Cura version they correspond to, but forgot this time.  I've updated the file names in the latest 2.7.2 release.  They only go up to Cura 5.6.0, but that will work with future Cura versions as well.

kartchnb, 2024-12-14:
Also, the Marketplace should now have version 2.7.2.